ros/nodelet_core #29: The wiki page has outdated bug tracker url

dirk-thomas, 2015-04-30:
http://wiki.ros.org/nodelet_core

Bug / feature tracker points to: https://code.ros.org/trac/ros-pkg/query?component=common&status=assigned&status=new&status=reopened

esteve, 2015-04-30:
I've updated the wiki page to use the `PackageHeader` macro and updated the `package.xml` file in each branch to point to the correct bugtracker.

esteve, 2015-04-30:
How long does it take for the wiki to be updated? Do I have to release a new version of `nodelet_core` for it to have effect?

dirk-thomas, 2015-04-30:
The `doc` job will index the repository every the days. Commonly the branch is registered in the distribution file, then you don't need a new release.

esteve, 2015-04-30:
@dirk-thomas Thanks. I'm closing the ticket now.

closed 2015-04-30 by esteve