ros/nodelet_core #32: header.seq is zero

antonioom, 2015-07-09:
I have written a small nodelet to save (on demand) the topics "depth/image_raw" and "depth/camera_info" to a rosbag. When I examine the rosbags later on, I notice that the header.seq field is always zero. This causes problems when I try to post-process the images stored in the rosbags with the image_proc and depth_proc packages. My guess is that header.seq doesn't get incremented when doing zero copy passing of data between nodelets. (The nodelet that produces the depth/image_raw and depth/camera_info topics is managed by the same nodelet manager that manages my nodelet.) If, however, I echo the header of either topic using an external node (e. g. rostopic echo ...) I do see the sequence number being increased. Shouldn't the header.seq get incremented also when zero copying data between nodelets?

tfoote, 2015-07-09:
Initially the seq number was specified to increment automatically when publishing. However there are conceptual problems with doing that inside the payload. (For example nodelets cannot do it with the zero copy methods as the data is read only, it could be republished again after being recieved if unchanged, and then the seq number could never be valid as two different subscribers. 

Likewise there are issues with the sequence number being publisher specific. There were some discussions on code.ros.org a long time ago about removing the seq because common behavior could not be relied upon. It was left in place with the current inconstant behavior to be non-breaking. 

There's a discussion ongoing for removing it in ROS 2.0: https://github.com/ros2/common_interfaces/pull/2 I don't think that there's anything we can do for this. 

image_proc and depth_image_proc should not be paying attention to the seq value. A quick search suggests that they are not: https://github.com/ros-perception/image_pipeline/search?utf8=%E2%9C%93&q=seq as I only see one function with that variable name and it's in a completely different context.

antonioom, 2015-07-16:
I was getting the error "Topics xxx/depth/image_raw and /xxx/depth/camera_info do not appear to be synchronized" and thought it had something to do with the seq numbers being all zero. I don't know what was causing the problem and unfortunately I can't reproduce it again. 

On a side note, I noticed that openni2_camera publishes a common depth/camera_info topic for both depth/image_raw and depth/image. This causes the seq number in its header to increment twice as fast as the seq of the other two topics, such that synchronization based on seq number wouldn't work. I also noticed that if I subscribe the image topic of a camera_publisher, the seq number of both the image topic and the info topic get incremented. However, if I subscribe to the info topic only, the seq number of the image topic does not get incremented. More arguments in favor of dropping the use of seq numbers?

Anyways, thanks for the quick response!

closed 2015-07-16 by antonioom