ros/nodelet_core #35: When set system time ahead nodelet crash

zuojiang518, 2015-12-22:
When set the system time ahead to one minute in the future, the nodelet crashed, and print as follow:
[ INFO] [1450779353.034638293]: Bond broken, exiting
[yocs_cmd_vel_mux-2] process has finished cleanly

Is this issue a bug?

tfoote, 2015-12-22:
Can you clarify that you jumped your system time forward by a minute while running? If so I would expect that would cause problems with bond, which is checking that you have connectivity via a heartbeat. If you jump a process forward by a minute all the timeouts for the heartbeat will expire and it will assume that it's been a full minute since it heard from the other end and nodelets are designed to shutdown when the bond is broken.

zuojiang518, 2015-12-26:
Yes, you're right, the reason is what you said. Because my device does not save time, so while it connected to network it will synchronize its system time with NTP. So the system time jumped, and the nodelets bond broken. I modified the judgement of timeout to continuous lost three heatbeats. Now when its system time changed the nodelets do not bond broken.

closed 2015-12-28 by zuojiang518

flixr, 2017-02-24:
I also just ran into this problem when using PTP to sync time...
I'm wondering why we don't "simply" use CLOCK_MONOTONIC to check the bond timeouts? Then this would not occur.

tfoote, 2017-02-24:
@mikaelarguedas FYI related to our discussions about bond

mikaelarguedas, 2017-02-24:
Thanks @flixr for reporting the issue. We were discussing this with @tfoote recently. I'll open an more detailed issue on the bond_core repo explaining the drawbacks of the different approaches and propose a solution.

flixr, 2017-02-24:
Thanks for the prompt feedback!
I'd be very much interested in a proposal/solution for this and can definitely offer help and testing for this here...

I also saw the issue https://github.com/ros/roscpp_core/issues/33 
Please note that I'm in no way endorsing the use of the monotonic clock for wall time or anything like that.
But IMHO the durations should use the monotonic clock by default (if not using sim time).
That is also slowly adjusted by NTP (and the likes), but doesn't have the jumps... so it seems perfect to me to use for durations, timeouts, etc...

Also note that basically all of our pipeline can deal with jumping time, but things like the bond timeouts still make it currently impossible to use in a real/production setup.

mikaelarguedas, 2017-02-24:
@flixr I started an issue here open for discussion: https://github.com/ros/bond_core/issues/16
I'll also have a look at the roscpp issue you pointed at.

I do agree that NTP and friends allow to slowly adjust time. It works for a lot of cases but I can imagine cases (e.g. SBC without battery for RTC) where you just can't drift slowly from epoch to "now" in a decent amount of time. So I think that being robust to time jumps would still be better than working around it by using monotonic clock.

flixr, 2017-02-25:
Don't get me wrong, we need something here that can deal with time jumps.
Just saying that CLOCK_MONOTONIC is also slowly slew adjusted by NTP, just doesn't jump.... so it's exactly what we want to compute timeouts.