ros/nodelet_core #36: Unit Testing Nodelets

progtologist, 2015-12-31:
While using nodelets, I discovered that the API does not provide a method for returning a nodelet object from the nodelet loader. Although such a method is not the proposed way to use nodelets, it would be very useful for unit testing nodelets with google tests. 

Right now, the only way to do this is to use the
```cpp
nodelet::Loader::Loader(const boost::function< boost::shared_ptr< Nodelet >(const std::string &lookup_name)> & 	create_instance)
```
method to instantiate the object, and to share that pointer with the unit tests. However it would be more elegant to be able to get the object (pointer) using a loader method.

daniel-dsouza, 2021-03-22:
do you by any chance still have an example of this?

progtologist, 2021-03-23:
Unfortunately I don't have access to that code anymore.
If I remember correctly though, you must create a function that creates nodelets and create a loader providing that function.
In your function you can have a std::map or even an std::vector that, while instantiating the nodelet, also adds it to that structure.
From your app, you use that new loader to instantiate the nodelet, and you can access that structure to get your instance at any time.

closed 2025-05-31 by sloretz