ros/nodelet_core #46: Race condition between waitForService and nodelet registration

asymingt, 2016-09-08:
Consider a scenario with two nodelets running within the same nodelet manager. One nodelet offers a ROS service which the other nodelet calls. There is no guarantee which service will start up first, so the logical step is to use the ```waitForService``` call in the client's ```Nodelet::onInit()``` implementation. For example:

```c++
#include <ros/ros.h>
#include <nodelet/nodelet.h>
#include <pluginlib/class_list_macros.h>
#include <example/Sample.h>

namespace example
{
  class Client : public nodelet::Nodelet
  {
  public:
    virtual void onInit() {
      example::Sample srv;
      srv_client_ = getNodeHandle().serviceClient<example::Sample>("/sample");
      if (!srv_client_.waitForExistence(ros::Duration(10.0))) {
        ROS_ERROR("Failed to find sample service");
        return;
    }
  private:
    // For registration
    ros::ServiceClient srv_client_;
  };
}

PLUGINLIB_EXPORT_CLASS(example::Client, nodelet::Nodelet) 
```

Unfortunately, the nodelet package uses a service call to register nodelets with ther manager. As a consequence there appears to be a race condition between nodelet registration and the ```waitForService``` call on custom services.

I don't think I'm the first to pick up this issue: [see this discussion](http://ros-users.122217.n3.nabble.com/Nodelets-td2002048.html).

Here's how to repeat this issue:

```
git clone https://bitbucket.org/asymingt/ros_issue
cd ros_issue/src
catkin_init_workspace
cd ..
catkin_make
source devel/setup.bash
roslaunch example example.launch
```

Keep retrying the last line and you will eventually find that the client node fails to find the service, even though there is a 10 second time out.

tfoote, 2016-09-08:
This is known expected behavior. 

From the main threading model documentation: http://wiki.ros.org/nodelet#onInit `This method is called on init and should not block or do significant work.`

All of the onInit calls are called from the same management thread sequentially. If you block inside the onInit of one nodelet, any subsequent nodelets will never initialize because you are blocking the thread that would initialize them and consequently you will timeout with any length of timeout. 

The easiest solution is to create a dedicated thread in the onInit which will just do your one shot execution and finishes. Though it's a little fragile to have a client that runs once at startup and does not have any logic to be retriggered at a later time say if settings change. Thus the startup case is simply that the settings have changed from unset to some value.

asymingt, 2016-09-08:
Thanks for explaining why this is the case. Do you know of any packages that use the approach you have suggested, so that I have some sort of threading pattern to follow? 

Right now I figured out that I can use a ROS timer to defer the execution of my initialization code by a second. This essentially permits nodelet registrations to take place before services are probed. My intuition is that this helps the problem, but does not completely solve it -- as it implicitly makes the assumption that nodelets are never loaded after the system is brought online, and that all nodelet registration completes within a second. 

I'd much prefer to use your approach. Any change that you can modify my brief example above to show the correct method of setting up a thread.

tfoote, 2016-09-08:
You can set the timer length to approximately zero and then use your waiting logic. The trick is that registering the timer gets you into the separate thread from the callback queue.

Add a private method:
```
void CallService() { 
//do my initialization here
}
```
And a private boost_thread class member:
`boost::thread& startup_thread_;`

And in the onInit
```
void onInit(){
startup_thread_ = new boost::thread(&ClassName::CallService, *this);
}
```

And for cleanliness you should probably also join the thread in the destructor.

asymingt, 2016-09-08:
Great, thanks! Didn't realise it was that simple.

closed 2016-09-08 by asymingt

dabonnie, 2021-02-11:
@tfoote is the nodelet manager OnInit() sequential behavior still the case for a multi-threaded environment? Or is OnInit() single threaded regardless?

tfoote, 2021-02-11:
Yes, it's the protection of the list of the nodelets that requires the protection not the number of threads available.