ros/nodelet_core #52: Creating kinetic-devel branch?

wkentaro, 2016-12-02:
@mikaelarguedas You stopped releasing new features to indigo and jade, so it is the time to create the kinetic-devel branch, right?

mikaelarguedas, 2016-12-02:
@wkentaro Can you be more specific about the features you are referring to ?

1.9.8 has been released for all distros (c.f status pages for [kinetic](http://repositories.ros.org/status_page/ros_kinetic_default.html?q=nodelet), [jade](http://repositories.ros.org/status_page/ros_jade_default.html?q=nodelet) and [indigo](http://repositories.ros.org/status_page/ros_indigo_default.html?q=nodelet))

I'm sometimes delaying releases on indigo and jade to allow more time for testing to be sure not to break anything on the previous distros but the goal for now is to avoid branching out as long as there are not big changes preventing to do so.

wkentaro, 2016-12-02:
I'd like to use this:
https://github.com/ros/nodelet_core/commit/e5cb1982fbb73e8b2887c2b5740d0d11fd5cf78a

But seeing ros/rosdistro, it does not seem released because 1.9.6-0 is closed:
https://github.com/ros/nodelet_core/commit/e5cb1982fbb73e8b2887c2b5740d0d11fd5cf78a

wkentaro, 2016-12-02:
@mikaelarguedas 
Oops, sorry, maybe I'm missing something.

I'm using shadow-fixed for apt and installed nodelet via apt,
and found the nodelet_lazy.h file as shown in below:

```
% source /opt/ros/indigo/setup.zsh

% rosversion nodelet
1.9.8

% ls /opt/ros/indigo/include/nodelet_topic_tools/nodelet_lazy.h
/opt/ros/indigo/include/nodelet_topic_tools/nodelet_lazy.h
```

mikaelarguedas, 2016-12-02:
1.9.6 has not been release because the timing between the indigo sync and the testing of these changes was not optimal. I closed it in favor of 1.9.7 that includes both the changes of #45 and #49. So you should be able to leverage both.
Thanks for both these contributions BTW :+1:

closed 2016-12-02 by wkentaro