ros/nodelet_core #71: does this tool support "relay" mode?

asimay, 2017-11-25:
does this tool support "relay" mode?
if not, and can you add it?

mikaelarguedas, 2017-11-25:
Hi @asimay,
Could you expllain what you mean by "relay" mode? Do you want to create node/nodelet that subscribe to a topic and republishes the same data on another topic ?

asimay, 2017-11-26:
hi, @mikaelarguedas , dear, yes. I used topic_tools before, and find your tools, topic_tools has "relay" mode, and I want to know your package has this kind of mode or not? because I need to defined this kind of mode across files.

mikaelarguedas, 2017-11-26:
Thanks for clarifying.
Currently `nodelet_topic_tools` does **not** provide a `relay` nodelet.

It could be fairly easily implemented by providing a nodelet templated on the message type and taking input and output topic names as parameters (you can look at nodelet_mux.h or nodelet_throttle.h for examples of nodelets templated on message types.

Though it is recommended to remap topics (if it makes sense in the case of your application) rather than subscribing and republishing them, this allow to avoid serialization/deserialization put as well as unnecessary copies. Please see https://github.com/ros/roscpp_core/issues/31 for more information of cases where copies will be performed anyway.

peci1, 2023-08-16:
http://wiki.ros.org/cras_topic_tools provides a relay nodelet. The package is released, so simply `sudo apt install ros-${ROS_DISTRO}-cras-topic-tools`.