ros/nodelet_core #76: add `-std=c++0x` manually

YvesZHI, 2018-05-25:
I'm working on Ubuntu 16.04.3 and the version of gcc is 5.4.0.

After downloading the ros package nodelet from Github and all of its necessary dependencies, I tried to execute `./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release` to compile it but failed.

Here is the detail: https://answers.ros.org/question/292165/nodelet-error-nullptr-was-not-declared-in-this-scope/

To compile it, I have to add `SET(CMAKE_CXX_FLAGS "-std=c++0x")` manually into the `CMakeLists.txt` files of `nodelet`, `test_nodelet` and `test_nodelet_topic_tools`.

gavanderhoorn, 2018-05-25:
@YvesZHI: it would be good if you could describe the platform that you are running Ubuntu 16.04.3 on. From your other questions on ROS Answers I suspect it may not be a regular `amd64` PC.

mikaelarguedas, 2018-05-25:
Hi @YvesZHI, thanks for reporting the issue.

Unfortunately this  doesnt not provide enough information to help you.
Providing the details of what platform, ROS distribution and branches of the repository you compile would be valuable.

Looking at the error you posted on ROS Answers, my guess is that you compiled the `melodic-devel` version of `class_loader` instead of the`indigo-devel` one.
To know what platforms are targeted by a ROS distribution, please refer to [REP3](http://www.ros.org/reps/rep-0003.html#melodic-morenia-may-2018-may-2023)
ROS Melodic targets only platforms with compilers using C++14 as their default. That is not the case for GCC5.4.

YvesZHI, 2018-05-26:
@mikaelarguedas 
Oh, you are right. After switching to the branch `indigo-devel` for the `class_loader`, everything works fine now.

closed 2018-05-26 by YvesZHI