ros/nodelet_core #77: getMTNodeHandle doesn't result in more callbacks in parallel

lucasw, 2018-07-28:
From https://answers.ros.org/question/298750/getmtnodehandlegetmtprivatenodehandle-doesnt-result-in-additional-callbacks-but-does-take-more-cpu/

I assumed from the line "getMTNodeHandle() and getMTPrivateNodeHandle() callbacks will be distributed over the thread pool of the manager" from http://wiki.ros.org/nodelet#Threading_Model that switching a subscriber in a nodelet from using getNodeHandle() would result in callbacks on a single topic distributed over threads available in the nodelet manager pool and would result in a higher callback rate when a single thread can't keep up.

What I've found in https://github.com/lucasw/nodelet_demo is that there are no additional callbacks, but I can see a huge increase in cpu.

Does a single topic callback still only get on thread, but multiple callbacks for multiple topics could take as many threads as there are topics?  I'll test this out.

Are there additional steps to make a nodelet multi-threaded?

A possible limited workaround would be to have a generic demultiplexer nodelet (which ought to be nearly zero cost with no copies or deserializations) and multiple instances of the nodelet subscribing to 1/n of the upstream topic rate.  Or if the above comment about a given topic being single thread is true, a single nodelet could have multiple subscribers to the multiple topics that the demux distributed.

drjsmith, 2021-09-22:
By default, concurrent callbacks by the same Subscriber are not allowed. However, you can enable them by setting the `allow_concurrent_callbacks` field of the [SubscribeOptions](http://docs.ros.org/en/indigo/api/roscpp/html/structros_1_1SubscribeOptions.html#ad6bc756dc2e7971110d1efa37d8c60e0.) to `true`.

[This issue](https://github.com/ros/ros_comm/issues/1545) explains the increase in cpu.

lucasw, 2021-09-23:
Thanks, I got that working like this and I can see the concurrency happening correctly:

```
ros::Subscriber sub1_;
...
ros::SubscribeOptions so1;
so1.init<std_msgs::Float32>(
      "input1",
      input_queue_size,
      boost::bind(&NodeletDemo::callback, this, _1));

so1.allow_concurrent_callbacks = true;

sub1_ = nh.subscribe(so1);
```
https://github.com/lucasw/nodelet_demo/blob/6160a2f28b351ee4da76ed48111a2927cb05972f/nodelet_demo/src/nodelet_demo.cpp#L127

closed 2021-09-23 by lucasw