ros/nodelet_core #87: Static linking to uuid which might be GPL?

130s, 2019-03-08:
This is similar post with https://github.com/ros/bond_core/issues/47. While nodelet depends on bondcpp, I open a separate ticket as I'm not certain if the discussion in this ticket is the same as said ticket.

INAL and I'm not entirely sure the following concern is valid.

`nodelet` [statically links to uuid](https://github.com/ros/nodelet_core/blob/2ec24ff596ca5d49295c85d951aa3b536ff10f8b/nodelet/CMakeLists.txt#L48), and `nodelet.cpp` [includes a header uuid.h](https://github.com/ros/nodelet_core/blob/2ec24ff596ca5d49295c85d951aa3b536ff10f8b/nodelet/src/nodelet.cpp#L46). 

On Ubuntu [rosdep key uuid](https://github.com/ros/rosdistro/blob/2540e78253740ba105fcf8d205d20be5f9855f9d/rosdep/base.yaml#L5101) translates to [uuid-dev](https://packages.ubuntu.com/xenial/uuid-dev). Content of `/usr/share/doc/uuid-dev/copyright` shows large part of the package is licensed GPL-2+. I haven't found uuid-dev's repo nor website, but [package info on debian.org](https://metadata.ftp-master.debian.org/changelogs/main/u/util-linux/util-linux_2.33.1-0.1_copyright) also lists the similar copyright file where many portions indicates GPL-2+.

Ins't bondcpp considered to be a derived work, and doesn't GPL force the derived work to be declared as GPL?

tfoote, 2019-03-08:
Viewing the contained copyright file in the debian package: http://changelogs.ubuntu.com/changelogs/pool/main/u/util-linux/util-linux_2.27.1-6ubuntu3/copyright

There are several licenses for different parts of that package. 

libuuid from that package is BSD licensed:
```
Files: libuuid/*
       libuuid/src/*
       libuuid/man/*
Copyright: 1996, 1997, 1998, 1999, 2007 Theodore Ts'o.
           1999 Andreas Dilger (adilger@enel.ucalgary.ca)
License: BSD-3-clause
```

This includes the header file you mention is used and the source for the library linked against.

shaun-edwards, 2019-03-10:
@130s, I have verified the link/license that to which @tfoote is linking.  I'm in agreement with his assessment.

130s, 2019-03-11:
Thanks for finding @tfoote. Without much experience in this regard, I agree with your finding to take the release note / changelog as the latest correct license info than the files being installed via the released binary software in my finding.

closed 2019-03-11 by 130s