ros/nodelet_core #94: cannot chain two of the same nodelet in the same nodelet manager

daniel-dsouza, 2019-08-30:
Hello,

I am trying to run two of the same nodelets in the same nodelet manager, and am having trouble remapping topics. If I run the nodelets as standalone, I can pass data from p1 to p2 to p3, but if I have both nodelets in the same manager, then crop1 still publishes data, but does not print any output with `NODELET_INFO_STREAM`, and crop2 does not seem to subscribe to the correct topic.

I am pretty sure that the nodelet code is not a fault, as the problem still occurs if I remove all of the logic and simply republish the subscribed point cloud.

test.launch
```xml
<?xml version="1.0" encoding="UTF-8" standalone="yes" ?>
<launch>
     <node pkg="nodelet" type="nodelet" name="lidar_nodelet_manager" args="manager" >
      <param name="num_worker_threads" value="8" />
    </node>

    <node name="crop1" pkg="swri_nodelet" type="nodelet" args="pcl_utils/crop_rectangular_box lidar_nodelet_manager">
      <!-- skip parameters -->
      <remap from="pointcloud_in" to="p1" />
      <remap from="pointcloud_out" to="p2" />
    </node>
    <node name="crop2" pkg="swri_nodelet" type="nodelet" args="pcl_utils/crop_rectangular_box lidar_nodelet_manager">
      <!-- skip parameters -->
      <remap from="pointcloud_in" to="p2" />
      <remap from="pointcloud_out" to="p3" />
    </node>
</launch>
```

klaxalk, 2021-01-28:
Hey, I guess that you already solved it since 2019, but it might be worth answering for the sake of others. I might say that the nodelet might be at fault. To truly remap topics you should remap them relatively to `crop1` and `crop2`, as:
```xml
    <node name="crop1" pkg="nodelet" type="nodelet" args="pcl_utils/crop_rectangular_box lidar_nodelet_manager">
      <!-- skip parameters -->
      <remap from="~pointcloud_in" to="~p1" />
      <remap from="~pointcloud_out" to="~p2" />
    </node>
    
    <node name="crop2" pkg="nodelet" type="nodelet" args="pcl_utils/crop_rectangular_box lidar_nodelet_manager">
      <!-- skip parameters -->
      <remap from="~pointcloud_in" to="/crop1/p2" />
      <remap from="~pointcloud_out" to="~p3" />
    </node>
```

But I suspect that those `pointcloud_in` and `pointcloud_out` might be coming out as `/pointcloud_in` and `/pointcloud_in`, which would suggest that you don't use a private node handler. I recommend the following within your nodelet:
```cpp
ros::NodeHandle nh_ = nodelet::Nodelet::getMTPrivateNodeHandle();
...
publisher_ = nh_.advertise<Typ>("topic_out", 10);
```

daniel-dsouza, 2021-01-29:
@klaxalk I have not seen the "~" in a remap before. Is it explicitly specifying that the topic is in the node(let) namespace?

klaxalk, 2021-01-29:
> @klaxalk I have not seen the "~" in a remap before. Is it explicitly specifying that the topic is in the node(let) namespace?

Yes, `~pointcloud_in` under the **crop1** node will actually remap `/crop1/pointcloud_in`. And with the private node handler, the topic (in my example `topic_out`) will come out `/crop1/topic_out` as unremapped. So this behavior will allow to run multiple instances of the same nodelet and/or nest them in groups.

daniel-dsouza, 2021-01-29:
Wonderful. While this may not solve the underlying issue, I think that your solution fixes the problem.

closed 2021-01-29 by daniel-dsouza

klaxalk, 2021-01-29:
Ah, and regarding the prints. When you run notelets under a nodelet manager, not as standalone, all the output is actually coming out of the nodelet manager. So when you run some nodelet in a separate terminal, don't look for its prints in its window, rather in the window with the manager.