ros/nodelet_core #99: Loader crash issue

ohryshchuk, 2019-11-04:
Hi,

There is a crash issue inside LoaderROS:

```
#0  0x00007fe209c44258 in std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, void*>, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, void*> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, void*> > >::find(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /opt/ros/kinetic/lib/libnodeletlib.so
No symbol table info available.
#1  0x00007fe209c3bf65 in nodelet::Loader::unload(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /opt/ros/kinetic/lib/libnodeletlib.so
No symbol table info available.
#2  0x00007fe209c4715b in nodelet::LoaderROS::unload(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /opt/ros/kinetic/lib/libnodeletlib.so
No symbol table info available.
#3  0x00007fe2097f2141 in bond::Bond::flushPendingCallbacks() () from /opt/ros/kinetic/lib/libbondcpp.so
No symbol table info available.
#4  0x00007fe2097f2675 in bond::Bond::breakBond() () from /opt/ros/kinetic/lib/libbondcpp.so
No symbol table info available.
#5  0x00007fe2097f3fc5 in bond::Bond::~Bond() () from /opt/ros/kinetic/lib/libbondcpp.so
No symbol table info available.
#6  0x00007fe209c52971 in nodelet::LoaderROS::~LoaderROS() () from /opt/ros/kinetic/lib/libnodeletlib.so
No symbol table info available.
#7  0x00007fe209c52af2 in boost::detail::sp_counted_impl_p<nodelet::LoaderROS>::dispose() () from /opt/ros/kinetic/lib/libnodeletlib.so
No symbol table info available.
#8  0x00007fe209c3a6fa in ?? () from /opt/ros/kinetic/lib/libnodeletlib.so
No symbol table info available.
#9  0x00007fe209c3abd5 in nodelet::Loader::~Loader() () from /opt/ros/kinetic/lib/libnodeletlib.so

```

Please consider fix in attachment. It addresses two issues:
1. Protection of bond_map_ deletion against concurrent access from multiple threads.
2. nodelets_ must outlive LoaderROS. According to above stack trace Loader accesses nodelets_ when it is already deleted.

[crash_fix.txt](https://github.com/ros/nodelet_core/files/3803469/crash_fix.txt)

vshum, 2019-11-14:
This crash was reproduced on Ubuntu 16.04.6 with latest ROS Kinetic Kame packages and nodelet 1.9.14

flixr, 2020-01-12:
@ohryshchuk please consider opening a PR for the fix