ros/nodelet_core #100: Cannot load nodelet /Plus for one exists with that name already

qwvLJV7C, 2019-12-16:
**distro**: Ubuntu bionic
**rosdistro**: melodic

I experimented with the --respawn and --no-bond flags basing on the http://wiki.ros.org/nodelet_tutorial_math and have a following issue: 

If the "respawn" option is set to a nodelet, e.g. "Plus", and then it gets crashed by e.g. sending a SIGSEGV signal, the following msg appears in a loop:

```
[ERROR] [1576491345.674093751]: Cannot load nodelet /Plus for one exists with that name already
[FATAL] [1576491345.674229784]: Failed to load nodelet '/Plus` of type `nodelet_tutorial_math/Plus` to manager `standalone_nodelet'
```

Sometimes it vanishes after some time, and the "Plus" nodelet process is restarted correctly, but sometimes it goes into an infinite loop.

The full **modified** launchfile from the http://wiki.ros.org/nodelet_tutorial_math below:

```
<launch>
  <node pkg="nodelet" type="nodelet" name="standalone_nodelet"  args="manager --no-bond" output="screen" respawn="true"/>

  <node pkg="nodelet" type="nodelet" name="Plus" args="load nodelet_tutorial_math/Plus standalone_nodelet --no-bond" output="screen" respawn="true">
    <remap from="/Plus/out" to="remapped_output"/>
  </node>

  <rosparam param="Plus2" file="$(find nodelet_tutorial_math)/plus_default.yaml"/>
  <node pkg="nodelet" type="nodelet" name="Plus2" args="load nodelet_tutorial_math/Plus standalone_nodelet --no-bond" output="screen" required="true">
    <rosparam file="$(find nodelet_tutorial_math)/plus_default.yaml"/>
  </node>
  <node pkg="nodelet" type="nodelet" name="Plus3" args="standalone nodelet_tutorial_math/Plus --no-bond" output="screen">
    <param name="value" type="double" value="2.5"/>
    <remap from="Plus3/in" to="Plus2/out"/>
  </node>
</launch>
```

tfoote, 2020-01-09:
This looks like expected behavior. If you disable the bond functionality. The nodelet will not be unloaded when you hard kill the nodelet placeholder process. The actual functionality is running in the manager and only goes away if it looses the bond connection to the placeholder process. If you use respawn to try to respawn the placeholder process it won't be able to do so because the nodelet is still actively running in the manager. That's why the bond is implemented, so that if the placeholder process is killed the nodelet will automatically unload.

closed 2020-01-09 by tfoote

qwvLJV7C, 2020-01-13:
Ok, got it, so the cultprit was the --no-bond flag.
Does it ever make a sense to have --respawn && --no-bond in the nodelet then?

tfoote, 2020-01-14:
It doesn't make sense to use those two together. In general for most users I wouldn't recommend using `--no-bond`. It's primary use case is for highly lossy networks where the bond is more likely to break than the nodelet crashing. A much better solution is to launch the nodelet launcher on the same machine as the manager then the loopback interface can be used and the reliability of the bond is high.