ros/nodelet_core #106: [BUG] Noetic: execution of code in 'onInit()' after timers/subscribers init can break said timers/subscribers

klaxalk, 2020-11-29:
Since the first commit in between Noetic and Melodic: https://github.com/ros/nodelet_core/commit/5272c3462e7c7ebab9848ed943d5c926b1e75842, the callback queue can be broken by executing code in `onInit()` between subscriber/timer initialization and the end of `onInit()`. The commit in fact causes the very problem it was supposed to fix.

Initially, I thought my problems were **ros:Timer**-related. So I submitted https://github.com/ros/ros_comm/issues/2085. But it is in fact much simpler and related to https://github.com/ros/nodelet_core/commit/5272c3462e7c7ebab9848ed943d5c926b1e75842.

- Any code with slightly longer execution (sleep(), few hundred ms loops, **Dynamic Reconfigure Server init**, etc) called after initializing subscribers/timer in `onInit()` breaks the callback queue.
- It affects more timers and subscribers with a high rate (>= 100 Hz).
- The broken timers and subscribers are broken for good.
- Subscribers break only if the subscribed topic offers messages during `onInit()` execution.

I think a user should be able to run a few-hundred-millisecond calculation in `onInit()`. If not, then this behavior is still very non-deterministic and undefined. It caused me a month full of headaches during which I reimplemented custom timer lib just to be able to move to Noetic, only later to realize it affects subscribers as well. Also, I think re-ordering things within `onInit()` is not a general solution (e.g., when using plugin lib plugins in your nodelet).

# Minimum breaking example

https://github.com/klaxalk/ros_timer_drs_bug

- `roslaunch timer_tester timer_tester.launch`
- to test the topics, call `rostopic pub /timer_tester/in std_msgs/Empty -r 100` while launching the node
- feel free to move the _subscriber init block_, the _DRS init block_, and the _dummy for loop_ within onInit().

Martin-Oehler, 2021-01-19:
This is a serious regression, which costed me a few hours as well (and certainly more if not for this issue, thank you @klaxalk). The offending commit should be reverted as soon as possible.

mjcarroll, 2021-01-26:
I have reverted this via #107 for now.  I would be willing to take another alternative if there is enough coverage to prove that it won't break existing use cases.

A new release has been cut for `noetic`, which should have a sync in a few weeks while this soaks in the testing buildfarm.  Look for the sync announcement on discourse.

https://github.com/ros/rosdistro/pull/28121

Apologies for the regression.

closed 2021-01-28 by klaxalk