ros/nodelet_core #108: Subscribing to a latched topic can cause indefinite hang in subscriber

mwcondino, 2021-02-11:
I encountered a very strange bug in `noetic` today. The gist of it is that if there's enough time between the creation of a subscriber (which is being published to by a latched pub) and exiting the `onInit` method, the latched message will never trigger a callback. Furthermore, this bug causes that subscriber to _never_ fire any callbacks.

To test this, I wrote a simple package implementing a tester class that simply subscribes to some topic upon initialization (with a sleep in the initializer). A timer is also registered to check if data was ever received on the input topic.

When a standard `node` executable is used, the message is received properly, but when the test class is loaded up from a nodelet, it shows that the message is never received.

Furthermore, I added a separate callback that allows the original subscriber to be re-initialized. When this happens, the nodelet is able to see the data finally!

Any ideas on why `nodelet` would behave differently than a standard `node` here? I've tried enabling debug logging, but haven't been able to find anything conclusive yet.

To test, unzip the attached package into a ROS workspace, then do `catkin_make`. The `test.launch` file spawns up a node and nodelet both wrapping the class `Tester`.

[tester.tar.gz](https://github.com/ros/nodelet_core/files/5968752/tester.tar.gz)

acarrillo, 2021-02-16:
Chiming in to say that I have experienced this exact symptom since switching from `kinetic` to `noetic` as well.

mwcondino, 2021-02-18:
Ah looks like this is likely a duplicate of https://github.com/ros/nodelet_core/issues/106 which has been patched in `1.10.1-1focal`. Just needed to upgrade my local version of `nodelet` and that fixed things.

closed 2021-02-18 by mwcondino