ros/nodelet_core #110: Nodelet hang on setParam

mark-goodall, 2021-06-16:
I'm not sure if this is the right place to report this issue, but if I have a nodelet (named for example `foo`) and there is a large amount of params in `/foo/*` When launching the nodelet it hangs/deadlocks with `[setParam] Failed to contact master at [localhost:11311].  Retrying...` I suspect that the length of the params is bigger than some maximum size and gets rejected. Possibly some more helpful warning or error message would be good here?

I traced it down to this line. The length of the xml was roughly about 60KB, and I solved it by cutting down this to a few KB.

https://github.com/ros/nodelet_core/blob/2d304e773bfa60b91bfae8ba2fc9d082d8015cd2/nodelet/src/nodelet.cpp#L224

jokla, 2021-08-23:
I have the same issue, but it is not happening all the time. Not sure what the problem is though.

peci1, 2021-09-08:
Maybe this is related? https://github.com/ros/ros_comm/issues/2182

mark-goodall, 2021-09-08:
Yes, this is the behaviour I am seeing. I've just tried this on a more recent ubuntu (which is also a different machine) and it doesn't happen. I'll close this issue, because it looks to be an issue elsewhere in ros.

closed 2021-09-08 by mark-goodall